cyberbotics/urdf2webots #11: Importing a Kinova Arm

jcoll44, 2019-01-03:
Hi,

I am trying to convert a URDF of a kinova arm into the PROTO format. I have the issue that when I import the PROTO into Webots all the joints are rotated at weird angles. The URDF I am converting is the m1n6s300_standalone.xacro from [kinova-ros](https://github.com/Kinovarobotics/kinova-ros) converted to URDF using ROS. See below an image of the broken file.

[image]

Any help in rectifying this would be very much appreciated.

Thanks

DavidMansolino, 2019-01-11:
Thank you for your bug report.
You are welcome to contribute if you want to fix this bug.
Otherwise, we will try to fix it as soon as we have time.

DavidMansolino, 2019-01-11:
In the meantime you can manually fix the axes issue following these steps:
  1. Add the robot in the world.
  2. Convert it to base node by right clicking on the robot in the scene-tree and pressing on 'Convert to Base Node(s)'.
  3. You can now directly edit the axes in the scene-tree.

If you want Cyberbotics to do this for you do not hesitate to contact support@cyberbotics.com

DavidMansolino, 2019-03-13:
Fixed in #12

closed 2019-03-13 by DavidMansolino