cyberbotics/urdf2webots #18: Fail to import the robomaster

OumarM, 2019-07-11:
I want to use the robomaster from the dji with my students.
We have tried to use the Gazebo model (https://github.com/jackychen227/ICRA2018_DJI_RM_AI_Challenge_NJUST) but I don't like it it doesn't look nice so I want to try it in the Weebots but when I try to import it the script seems blocked:
```
./urdf2webots.py --input=../robots_description/infantryb/robots/infantryb.urdf
Robot name: infantryb
```

What am I doing wrong?

We also want to use the simulation to train for the dji ai challenge (https://www.dji.com/newsroom/news/robomaster-2019-icra-ai-challenge), can you confirm this is possible?

OumarM

DavidMansolino, 2019-07-12:
Thank you for this bug report, I just tried and it seems you are not doing anything wrong but there is indeed a problem with the importer when used with this urdf file. We will try to fix this as soon as possible and let you know when it is done!

> We also want to use the simulation to train for the dji ai challenge (https://www.dji.com/newsroom/news/robomaster-2019-icra-ai-challenge), can you confirm this is possible?

Yes that's of course possible, and I am sure you will get a more realistic and nice looking result than in Gazebo :wink:

DavidMansolino, 2019-07-16:
There was indeed a performance issue (the script was taking hours to perform the conversion).
We have improved the situation in #19, and we also added a new `--disable-mesh-optimization` argument that removes some mesh optimization but speeds up a lot the conversion, in your case I strongly recommend to use this argument.