cyberbotics/urdf2webots #23: Error during conversion

SunnyKatyara, 2019-09-20:
I am trying to convert urdf of my robot into webots but its showing following error.

Root link: panda_virtual_link_0
Traceback (most recent call last):
  File "urdf2webots.py", line 138, in <module>
    parserURDF.parseGazeboElement(child, rootLink.name, linkList)
  File "/home/sunny/development/catkinws/src/urdf2webots/parserURDF.py", line 801, in parseGazeboElement
    print('Unsupported "%lf" image format, using "R8G8B8A8" instead.' % imageElement.getElementsByTagName('format')[0].firstChild.nodeValue)
TypeError: float argument required, not unicode

Can you please help me to convert it.

Thank You

DavidMansolino, 2019-09-20:
Hi, thank you for the bug report.
Which version of Python are you using?
would it be possible for you to share the URDF file?

SunnyKatyara, 2019-09-20:
I am using 2.7 and i do not know how to share mu URDF here.

DavidMansolino, 2019-09-20:
You simply have to put it in a zip archive and you can then drag and drop it on the text area.

SunnyKatyara, 2019-09-20:
[panda.tar.gz](https://github.com/cyberbotics/urdf2webots/files/3635941/panda.tar.gz)

Here is the URDF with all meshes.....

DavidMansolino, 2019-09-20:
Thank you, we will try it and let you know what is not working.

DavidMansolino, 2019-09-20:
I just found the cause of the issue, and I have created a patch here: #24 

I checked with your URDF file and it seems to work fine (raw import without any edition):
[image]

Note also that your URDF file contains some paths `package://tutorial_iros2018/robots/` please replace them by the actual path to your folder (e.g. search and replace in the file `package://tutorial_iros2018/robots/panda` by `home/username/panda`).

Let me know if it woks fine with this patch for you too.

SunnyKatyara, 2019-09-20:
You used urdf2webots converter? am i right? Just its the issue with path?

DavidMansolino, 2019-09-20:
Yes, I did use the urdf2webots convert.
Your import fails for two reasons:
  - It need the fix of #24 (to get it simply clone the repository on this branch: `git clone -b fix-error-message https://github.com/cyberbotics/urdf2webots.git`)
  - The paths need to be corrected.

SunnyKatyara, 2019-09-20:
Thank you. The code is working now. But i am unable locate my file. the output file. Where can i find it normally? Because in the Urdf2webots i can see panda_texture folder but no PORTO folder.

Thank You

SunnyKatyara, 2019-09-20:
Working. Thank You

DavidMansolino, 2019-09-20:
Good news!

It will create directly a panda.proto file. you can then copy this file in the `proto` folder of your project and then you will be able to add the robot to your simulation.

PR #24 (2019-09-20): Fix  Error Message Format Causing Failure.
Fix  #23
closed 2019-09-20 by DavidMansolino