cyberbotics/urdf2webots #25: Unable to able porto file into webots simulation

SunnyKatyara, 2019-09-20:
Hello,

I have converted my urdf into porto file but unable to import it.

1. In my project, my porto file is available in portos
2. Launching simulation with emoty.wbt
2. When i add with porto nodes, i am unable to see the file.

I do not what is missing ?

DavidMansolino, 2019-09-20:
If you did correctly add the `panda.proto` file to the `protos` folder you should be able to add it to the simulation like this:
[image]
If it is not working try to close and re-open Webots (it my require to restart to update the list of protos).

SunnyKatyara, 2019-09-20:
[image]
It is giving me this error

SunnyKatyara, 2019-09-20:
If possible can you please share with me that proto file that you created. I am new to webots and i really want to explore it for my research.

SunnyKatyara, 2019-09-21:
I tried but it is showing me errors all the time. Attached is the conversion data. 
1. It is saying can not determine package root path
2. Unsupported "R8G8B8" image format, using "R8G8B8A8" instead.

[image]
[image]

SunnyKatyara, 2019-09-23:
Hello, i can import other protos from webots library but the one that i create is showing me these errors. Can you please help me to figure it out.

Thank you

DavidMansolino, 2019-09-23:
> It is giving me this error

You can ignore these errors in a first step, they will not prevent you from using the proto.

> If possible can you please share with me that proto file that you created. I am new to webots and i really want to explore it for my research.

Here is the file: 
[Panda.proto.zip](https://github.com/cyberbotics/urdf2webots/files/3642107/Panda.proto.zip)

> It is saying can not determine package root path

This means that you did miss some file path update when you edited the URDF file. You should not have any more `package://tutorial_iros2018/...` in your URDF file:
You should do a search and replace for the `package://tutorial_iros2018/robots/` string and replace it by the actual path to your `panda` folder.

> Unsupported "R8G8B8" image format, using "R8G8B8A8" instead.

This message just say that the camera will use the `R8G8B8A8` image format instead of the `R8G8B8` one.

Let me know if you still have issues.

SunnyKatyara, 2019-09-23:
Thanks! Its working

DavidMansolino, 2019-09-23:
You're welcome :wink:

closed 2019-09-23 by DavidMansolino