cyberbotics/urdf2webots #27: Robot Falling down when simulation starts.

SunnyKatyara, 2019-09-23:
When i start the simulation,the robot falls down. Its probably the transformation. I fixed it in the world but still it falls out. What to do? I want to use to communicate with my robot to test the controller i developed with Gazebo. 

How to fix this issue.

Thank You

DavidMansolino, 2019-09-24:
It seems the bounding object of the robot is not correct (too big and not centered):

[image]

I would advise trying the following:
  - Re-import your robot with the `--box-collision` argument.
  - Edit manually the bounding object and center of mass of the Solid nodes of the robot to make sure it's center of mass is more or less centered.

If you are not familiar with Wetbos, I also strongly recommend following our tutorials: https://www.cyberbotics.com/doc/guide/tutorials

SunnyKatyara, 2019-09-24:
I already did it with box collision. I think its because of the orientation of my platform. When i import it, the robot is rotated about x-axis by -90 degrees and when i run the simulation i tries to move to 0 degress about x-axis. 

Moreover, i want to know that in webots, is it possible to control each of the joints like in gazebo via ROS. Because the examples that i saw with webots-ros have python script running as node but could not figure out the basis for control.

Looking forward to hear from you.

SunnyKatyara, 2019-09-24:
Can i use the same set of codes to control my robot in webots. Since, the rendering ability of webots is way more better than gazebo. I have designed few of my worlds in Blender and rendered them into Gazebo and now i want to do the same with webots. But before that i want to check that is compatible with all my files i wrote for gazebo. Please do confirm me. 

Thank You

DavidMansolino, 2019-09-25:
> I already did it with box collision. I think its because of the orientation of my platform. When i import it, the robot is rotated about x-axis by -90 degrees and when i run the simulation i tries to move to 0 degress about x-axis.

Yes, that is maybe the problem.

> Moreover, i want to know that in webots, is it possible to control each of the joints like in gazebo via ROS. Because the examples that i saw with webots-ros have python script running as node but could not figure out the basis for control.

Yes, the idea is to have one python node running as a bridge between Webots and ROS, then the actual control is done in your usual nodes (like you do with real robot and Gazebo).
We have developed one of this node that is very generic (it is the 'ros' controller) but it is not the same services and topic that the one used by Gazebo or the real robots: https://cyberbotics.com/doc/guide/tutorial-8-using-ros
We are improving the situation to create better ROS interface to have the exact same interface than with gazebo and the real robot.
We have done this for the Universal Robots robotic arms: https://github.com/cyberbotics/webots/tree/revision/projects/robots/universal_robots/resources/ros_package/ur_e_webots
This is a good starting point for creating your own interface.

Or if you are using (or will) use ROS2, we are doing a lot of progress on this front and will soon have interfaces with many robot model using native services/topics/actions:
  - http://wiki.ros.org/webots_ros2
  - https://github.com/cyberbotics/webots_ros2

SunnyKatyara, 2019-09-25:
I even tried that one too but again it is falling down. It is probably due the reason that you have pointed out. For the solid objects in my proto how can i define these body masses and make sure that they are at center. Is there any way to do that? In terms of rendering, Webots is super cool, i really like it, so curious to explore it.

One more question i want ask you that, i have a environment designed in blender, is it possible to render it in the webots? or any way to import it in webots?

Thank You

DavidMansolino, 2019-09-26:
One simple solution to test this is to right click on your PROTO node in Webots and select 'Convert to Base Nodes' this will explode the PROTO and let you change the internal nodes (e.g. the Physics and BoundingObjects nodes).

> One more question i want ask you that, i have a environment designed in blender, is it possible to render it in the webots? or any way to import it in webots?

Yes of course, there are two options:
  - export your environment in *.wrl file from blender and them import the file in Webots from the `File / Import VRML97` menu.
  - use our blender exporter to directly generate the world file: https://github.com/cyberbotics/blender-webots-exporter

SunnyKatyara, 2019-09-27:
Thanks. I am able to import my world in the webots from the blender. But the problem with robot falling down is not resolved. I tried with base nodes too edit the physics and bounding objects but unable to figure out still what changes will fix this issue.

Thank You

closed 2019-10-18 by DavidMansolino