cyberbotics/urdf2webots #31: Feature request: custom proto code in urdf

Timple, 2019-10-22:
For our application we need support for closed-loop kinematics.
Unfortunately the urdf format does not support it. 

I could modify the generated .proto file to contain the [SolidReference](https://cyberbotics.com/doc/reference/solidreference) required to do so. But this would mean the proto starts deviating from the urdf and cannot be re-generated anymore.

Would it be possible / wise to allow for custom .proto entries in the urdf model? Kind of similar how you can add sdf elements for gazebo.

DavidMansolino, 2019-10-22:
That's feasible on the importer side this should be easy to support. But on the URDF side, this will differ from the norm and create non-standard URDF file, this might be problematic when you then want to use your URDF file with something else than the importer, right?

Timple, 2019-10-22:
Unknown elements seem to be ignored. Now we have items like this in our urdf:
```xml
<gazebo reference="left_track_link">
  <material>Gazebo/Black</material>
</gazebo>
```
And I just tested if I created a <webots> element and nobody complained :).

Ideally it can reference links/joints so we can keep the webots.urdf file seperate from the rest of the implementation.

DavidMansolino, 2019-10-22:
Ok, that makes sense indeed, adding a `<webots>` tag seems fine.
Then would something like this be fine?
```xml
    <link name="wrist_3_link">
      <visual>
        <geometry>
          <mesh filename="myfile.dae" />
        </geometry>
      </visual>
      <collision>
        <geometry>
          <mesh filename="myfile.stl" />
        </geometry>
      </collision>
      <webots>
        <proto name="myProtoName">
          <field name="translation" value="0 0 0"/>
          <field name="rotation" value="0 1 0 0"/>
        </proto>
      </webots>
    </link>
```

If you have any other ideas in mind do not hesitate to make some proposition.

Timple, 2019-10-22:
I think that is already a good proposal!
But it also 'clutters' the urdf file for normal operations.

In addition to the above this would also be nice:
```xml
    <link name="wrist_3_link">
      <visual>
        <geometry>
          <mesh filename="myfile.dae" />
        </geometry>
      </visual>
      <collision>
        <geometry>
          <mesh filename="myfile.stl" />
        </geometry>
      </collision>
    </link>

<!-- Somewhere else in an include file: -->
<webots reference="wrist_3_link">
  <node name="NodeName">
    <field name="translation" value="0 0 0"/>
    <field name="rotation" value="0 1 0 0"/>
  </proto>
</webots>
```