cyberbotics/urdf2webots #33: Feature request: generate separate .proto files for meshes

Vidicon, 2019-12-02:
Currently the script generates one big .proto file. For a complex robot this is can make the .proto file very large. Large files are difficult to edit as there are quit unclear and the text editor becomes very slow. 

In some example robots (kuka youbot) I have seen the robot been split into separate files. It would be very nice if the converter script would give you the option to generate a separate .proto files for the original mesh files.

Advantages:
- easier editing of the model.
- cleaner model files. 
- decrease single file size.
- mesh files become reusable. 
- clearer version control.

DavidMansolino, 2019-12-02:
If you import the PROTO in Webots you can then 'unprototize' it which makes the edition in Webots simpler, you simply have to right-click on the robot in Webots and select 'Convert to base Node(s)'.
This can be a solution, but of course, your proposal to create several PROTOs files is good too.

Vidicon, 2020-08-01:
Started working on this feature. 
Tested it using the turtlebot3_description, it seems to be working.

closed 2020-08-03 by Vidicon