cyberbotics/urdf2webots #41: Exporter issue with stl files

Acwok, 2020-03-04:
The exporter seems to triangulate the mesh, and sometimes it flips face normal (see picture).
This only happens with stl not with dae.
Also, the dae is smoothed (left most mesh in the picture) in contrary to the stl.
[image]

I've check the normals in blender, they're ok.
Here is the command I used: 
```
python -m urdf2webots.importer --input=robot.urdf --output=robot.proto --box-collision --disable-mesh-optimization
```

omichel, 2020-03-05:
About the smoothing, I would recommend you to edit the resulting Webots PROTO in a text editor and inside the [IndexedFaceSet](https://www.cyberbotics.com/doc/reference/indexedfaceset) nodes, set a `creaseAngle` value of about 3 or so. That will smooth the faces.

omichel, 2020-03-05:
It is not exactly that the normals are flipped, but rather that the triangle order is reversed (making the face appear on the wrong side). This is likely due to a wrong face in the source model. Blender by default displays both faces for triangles, so you may not be able to see the problem unless you disable double-sided face rendering in Blender.

Acwok, 2020-03-05:
I've double-checked the stl with blender (backface culling on) the faces seems fine.
Here is a simplified URDF using these stl. The proto generated from the URDF has a hole in the cylinder (see screenshot) while it hasn't when displayed with rviz.
[mesh_normal_issue.tar.gz](https://github.com/cyberbotics/urdf2webots/files/4293132/mesh_normal_issue.tar.gz)

[image]

omichel, 2020-03-06:
OK, I could investigate the problem and found that the triangle is actually simply missing (it's not a problem of wrong vertex order or normal). It could be that the `urdf2webots` converter has a little bug when converting STL meshes to `IndexedFaceSet` nodes and misses the last triangle or so. Could you try to export the same models from Blender in a different format, like DAE or OBJ instead of STL and see whether that fixes the problem?

Acwok, 2020-03-06:
The converter works well with dae.

omichel, 2020-03-06:
OK, so it's very likely a bug with STL import.

DavidMansolino, 2020-03-11:
There is definitely something wrong with this face in the original STL model, see how this face looks compared to the others once selected in Blender:
[image]

DavidMansolino, 2020-03-11:
Here is a fixed version of the mesh:
[link_0.stl.zip](https://github.com/cyberbotics/urdf2webots/files/4316996/link_0.stl.zip)

Acwok, 2020-03-11:
See `mesh_normal_issue.tar.gz` archive that should reproduce the issue.
I've updated the paths to the stl in the urdf.
Then, I've converted from urdf to proto using: `python -m urdf2webots.importer --input=/workspace/mesh_normal_issue/dlsaffe.urdf --output=/workspace/Dlsaffe.proto --disable-mesh-optimization`.
Finally, when opened in Webots, I still see the same issue:

[image]

---

By the way, I see the same number of faces in both meshes when importing the old and new stl in Blender.

> see how this face looks compared to the others once selected  

I think it's just the way blender indicates the previous selected face. It can be reproduced on the stl you sent. Unselect all > select a face > select all

DavidMansolino, 2020-03-11:
> I think it's just the way blender indicates the previous selected face. It can be reproduced on the stl you sent. Unselect all > select a face > select all

It seems you are right, I thought that it was indicating that something was wrong with this face (as it was matching with the problematic one), but you are right that it simply indicates the previously selected face.

BenjaminHug8, 2022-02-04:
I am closing this issue as the mesh files are not anymore converted by the `urdf2webots` tool but rather by the Assimp library when Webots parse worlds and protos.

Feel free to reopen this issue or create a new one in case you have difficulties to import your mesh files.

closed 2022-02-04 by BenjaminHug8