cyberbotics/urdf2webots #42: Joint axes not correctly exported

Acwok, 2020-03-04:
Some hingejoint axis doesn't correspond to the URDF after converting using `urdf2webots`.
For example, in Rviz (using `roslaunch urdf_tutorial display.launch model:=robot.urdf.xacro`) when joint_2 is increased, and doing the same in Webots, the axis of rotation is sometimes different (joint does not rotate in the same direction or joint does not rotate in the same axis).

See the following screenshot:
[image]

omichel, 2020-03-05:
Can you post here a simple URDF file that exhibits this problem?

Acwok, 2020-03-05:
Yes sure, here is the catkin package containing the urdf + mesh:
[dlsaffe_arm_rail_demo_pdr.tar.gz](https://github.com/cyberbotics/urdf2webots/files/4293048/dlsaffe_arm_rail_demo_pdr.tar.gz)

Here is the urdf with absolute path for the mesh:
[dlsaffe.urdf.tar.gz](https://github.com/cyberbotics/urdf2webots/files/4293065/dlsaffe.urdf.tar.gz)

And finally the proto converted from the urdf:
[Dlsaffe.proto.tar.gz](https://github.com/cyberbotics/urdf2webots/files/4293072/Dlsaffe.proto.tar.gz)
(the physics field of the first joint has been removed)

Simon-Steinmann, 2020-08-05:
I witnessed this issue several times now. Joint2 in this model shows this issue:

[Mirobot.zip](https://github.com/cyberbotics/urdf2webots/files/5027844/Mirobot.zip)

I think I found the issue. The axis of the Hingejoint is not rotated correctly, if the urdf joint has a origin rpy that is NOT 0 0 0

Simon-Steinmann, 2020-08-05:
I fixed it and created a PR

closed 2020-08-05 by Simon-Steinmann