cyberbotics/urdf2webots #44: PROTO and Filename missmatch

DavidMansolino, 2020-06-02:
In some cases it seems that the generated PROTO name and filename are not matching.
This is reproducible for example with the 'braccio' robot from here: https://github.com/jonabalzer/braccio_description

micsche, 2020-08-10:
Same problem here.

from Solidworks urdf export. I have 4 links and 3 joints. However I am only getting the root link exported into Proto.

python -m urdf2webots.importer --input=/home/mike/Documents/walker/urdf/walker.urdf --box-collision --multi-file
Robot name: walker
Parsing Mesh: /home/mike/Documents/walker/meshes/thigh.STL
Parsing Mesh: /home/mike/Documents/walker/meshes/femur.STL
Parsing Mesh: /home/mike/Documents/walker/meshes/leg.STL
Parsing Mesh: /home/mike/Documents/walker/meshes/pod.STL
Root link: pod
There are 4 links, 3 joints and 0 sensors
Create meshFile: podMesh.proto

Walker.proto file (below) and podMesh.proto created
#VRML_SIM R2020b utf8
# license: Apache License 2.0
# license url: http://www.apache.org/licenses/LICENSE-2.0
# This is a proto file for Webots for the Walker
# Extracted from: /home/mike/Documents/walker/urdf/walker.urdf

PROTO Walker [
  field  SFVec3f     translation     0 0 0
  field  SFRotation  rotation        0 1 0 0
  field  SFString    controller      "void" # Is `Robot.controller`.
  field  MFString    controllerArgs  []     # Is `Robot.controllerArgs`.
  field  SFString    customData      ""     # Is `Robot.customData`.
  field  SFBool      supervisor      FALSE  # Is `Robot.supervisor`.
  field  SFBool      synchronization TRUE   # Is `Robot.synchronization`.
  field  SFBool      selfCollision   FALSE  # Is `Robot.selfCollision`.
]
{
  Robot {
    translation IS translation
    rotation IS rotation
    controller IS controller
    controllerArgs IS controllerArgs
    customData IS customData
    supervisor IS supervisor
    synchronization IS synchronization
    selfCollision IS selfCollision
    children [
      podMesh {
      }
    ]
    name "pod"
    boundingObject Transform {
      translation 0.000000 0.014500 0.000000
      children [
        Box {
          size 0.079727 0.029000 0.080000
        }
      ]
    }
    physics Physics {
      density -1
      mass 0.046773
      centerOfMass [ 0.000328 0.004940 0.000082 ]
    }
  }
}

DavidMansolino, 2020-08-10:
Can you please share the URDF file so that we can try reproducing it?

micsche, 2020-08-10:
Robot urdf file structure uploaded here:

https://github.com/micsche/walker_robot

closed 2020-08-24 by DavidMansolino