cyberbotics/urdf2webots #57: Feature Request: Add Fields (collision, static base etc.)

Simon-Steinmann, 2020-08-05:
I converted several robotic arms, and there are a few things I always have to do manually:
1. Adding these fields
```

  field SFBool     supervisor      FALSE             # Is `Robot.supervisor`.
  field SFBool     synchronization TRUE              # Is `Robot.synchronization`.
  field SFBool     selfCollision   TRUE              # Is `Robot.selfCollision`.
  field MFNode     toolSlot        []                # Extend the robot with new nodes at the end of the arm.
  field SFBool     staticBase      TRUE              # Defines if the robot base should ba pinned to the static environment.
```

2. Adding the if statement to the Physics node of the base_link and implementing the Fields
 ```
   %{ if fields.staticBase.value == false then }%
    physics Physics {
      density -1
      mass 4
      centerOfMass [
        0 0 0
      ]
    }
    %{ end }%
    controller IS controller
    controllerArgs IS controllerArgs
    supervisor IS supervisor
    synchronization IS synchronization
    selfCollision IS selfCollision
  }
}
```

3. Adding the ToolSlot Transform in the last link:

```
                                      endPoint Solid {
                                        translation 0 0 0.12
                                        children [
                                          Shape {
                                            appearance USE METAL_APPEARANCE
                                            geometry IndexedFaceSet {
                                              coord Coordinate {
                                                point [... ]
                                              }
                                              coordIndex [...   ]
                                              creaseAngle 0.6
                                            }
                                          }
                                          Transform {
                                            translation 0 0.12 0
                                            children IS toolSlot
                                           }

```

It would be very beneficial, if these things could be added automatically (especially 1. and 2.)

DavidMansolino, 2020-08-05:
These fields should be very easy to add as they are generic:
```
  field SFBool     supervisor      FALSE            # Is `Robot.supervisor`.
  field SFBool     synchronization TRUE          # Is `Robot.synchronization`.
  field SFBool     selfCollision   TRUE              # Is `Robot.selfCollision`.
```

To be added here:
https://github.com/cyberbotics/urdf2webots/blob/c04de63e066e592dc8ff1b339ec00b4fb08d2a51/urdf2webots/writeProto.py#L56
And here:
https://github.com/cyberbotics/urdf2webots/blob/c04de63e066e592dc8ff1b339ec00b4fb08d2a51/urdf2webots/writeProto.py#L71

For the others, it might be a bit more complicated as they are quite specific to robotic arm.

Simon-Steinmann, 2020-08-06:
Perhaps there could be a flag '--robotic-arm' that can be set, adding the static base and toolslot. And perhaps adding the static base in general, with it being set to False as default. Many robots have static bases. This exact issue was one of the reasons that I didnt switch to webots a year ago from gazebo. I was trying to get the kinova jaco2 arm into webots, but had issues like the robot just falling over etc. It takes a while to understand how the scene tree, nodes, fields and especially proto files work.

closed 2020-08-07 by Simon-Steinmann