cyberbotics/urdf2webots #85: The option `--box-collision` of the convertor seems to work badly when convert a stl file

zapplelove, 2020-09-17:
I have a robot's urdf file, and the visual of the robot uses the stl file. I have converted it into webots using this tool, but when I check the model, I find that the inertia of the solid is not set. So I read the documents of the webots, it said that when give the boundingObject of the solid, the inertia will calculate automatically. So I use the tool again to re-generate the proto file to get a `box` like collision shape, but the option seems to work badly.
For example:
the shape of solid `Body` is :(Mass of it is 11kg, the shape is complex)
```
                        # Shape {
                        #   appearance PBRAppearance {
                        #     baseColor 0.5 0.5 0.5
                        #     transparency 0
                        #     roughness 1
                        #     metalness 0
                        #     emissiveColor 0 0 0
                        #   }
                        #   geometry IndexedFaceSet {
                        #     coord Coordinate {
                        #       point [
                        #         ...
                        #     ]
                        #       coordIndex[
                        #         ...
                        #     ]
                        #     creaseAngle 1
                        #   }
                        # }
```
the results of the conversion:
```
                      boundingObject Sphere {
                        radius 0.005
                      }
```
So, Should I specify the inertia of the robot's rods directly in Numbers? But how do I make sure that I'm assigning the right inertia? Do webots have visual options like Gazebo to see the shape of the inertia?

DavidMansolino, 2020-09-17:
The box collision is supposed to generate only boxes, I don't understand where does this 'Sphere' nodes come from?
Are you using the latest version of the importer from the master branch or the one from pip?
Can you share the original URDF file and the exact command you are using to convert it?

zapplelove, 2020-09-17:
> The box collision is supposed to generate only boxes, I don't understand where does this 'Sphere' nodes come from?
> Are you using the latest version of the importer from the master branch or the one from pip?
> Can you share the original URDF file and the exact command you are using to convert it?

Yes, it's a strange result.
I use a `virtualenv` and use the `pip install urdf2webots` to install the tool.
I use the command: `python -m urdf2webots.importer --input=cheetah.urdf --box-collision --output=cheetah.proto`
and the origin urdf file is :
[cheetah.txt](https://github.com/cyberbotics/urdf2webots/files/5239528/cheetah.txt)
and a stl file of it is:
[back.zip](https://github.com/cyberbotics/urdf2webots/files/5239551/back.zip)

DavidMansolino, 2020-09-17:
Can you maybe retry from the up to date files from the master branch of this repo instead of pip?
We fixed several issues about bounding object and inertia since the latest pip release.

zapplelove, 2020-09-17:
> Can you maybe retry from the up to date files from the master branch of this repo instead of pip?
> We fixed several issues about bounding object and inertia since the latest pip release.

Ok, I will try and reply soon.

DavidMansolino, 2020-09-17:
Perfect, thank you.

zapplelove, 2020-09-18:
I try the tool from the master branch. Here are the steps:
+ git clone ...
+ workon <python virtualenv>
+ source devel/setup.bash
+ python ~/urdf2webots/demo.py --input=cheetah.urdf --output=cheetah.proto --box-collision
here is the results:
```bash
Directory "cheetah_textures" already exists!
Parsing Mesh: <catkin_ws>src/cheetah/meshes/back.STL
Parsing Mesh: <catkin_ws>src/cheetah/meshes/link.STL
Parsing Mesh: <catkin_ws>src/cheetah/meshes/Vl2.STL
Parsing Mesh: <catkin_ws>src/cheetah/meshes/Hl2.STL
Parsing Mesh: <catkin_ws>src/cheetah/meshes/Sl2.STL
Parsing Mesh: <catkin_ws>src/cheetah/meshes/Vl1.STL
Parsing Mesh: <catkin_ws>src/cheetah/meshes/Hl1.STL
Parsing Mesh: <catkin_ws>src/cheetah/meshes/Sl1.STL
Parsing Mesh: <catkin_ws>src/cheetah/meshes/Body.STL
Warning: This mesh has a lot of triangles!
Warning: It is recommended to use the script with the '--disable-mesh-optimization' argument.
Parsing Mesh: <catkin_ws>src/cheetah/meshes/FS.STL
Warning: This mesh has a lot of triangles!
Warning: It is recommended to use the script with the '--disable-mesh-optimization' argument.
Parsing Mesh: <catkin_ws>src/cheetah/meshes/L1.STL
Parsing Mesh: <catkin_ws>src/cheetah/meshes/L2.STL
Parsing Mesh: <catkin_ws>src/cheetah/meshes/L3.STL
Parsing Mesh: <catkin_ws>src/cheetah/meshes/Hb.STL
Parsing Mesh: <catkin_ws>src/cheetah/meshes/Vb.STL
Parsing Mesh: <catkin_ws>src/cheetah/meshes/FSb.STL
Parsing Mesh: <catkin_ws>src/cheetah/meshes/RS.STL
Warning: This mesh has a lot of triangles!
Warning: It is recommended to use the script with the '--disable-mesh-optimization' argument.
Parsing Mesh: <catkin_ws>src/cheetah/meshes/RSb.STL
Parsing Mesh: <catkin_ws>src/cheetah/meshes/L4.STL
Root link: world
There are 52 links, 51 joints and 2 sensors
```
and the error still is there:cold_sweat: , some `boundingObject`s  still are `Sphere`s
hare are screenshots of the generated proto file:
[image]
[image]
[image]

The good news is that inertiaMatrix is being generated, though I don't know if it's generating the right results

zapplelove, 2020-09-18:
> I try the tool from the master branch. Here are the steps:
> 
> * git clone ...
> * workon `<python virtualenv>`
> * source devel/setup.bash
> * python ~/urdf2webots/demo.py --input=cheetah.urdf --output=cheetah.proto --box-collision
> 
> here are the results:
> 
> ```shell
> Directory "cheetah_textures" already exists!
> Parsing Mesh: <catkin_ws>src/cheetah/meshes/back.STL
> Parsing Mesh: <catkin_ws>src/cheetah/meshes/link.STL
> Parsing Mesh: <catkin_ws>src/cheetah/meshes/Vl2.STL
> Parsing Mesh: <catkin_ws>src/cheetah/meshes/Hl2.STL
> Parsing Mesh: <catkin_ws>src/cheetah/meshes/Sl2.STL
> Parsing Mesh: <catkin_ws>src/cheetah/meshes/Vl1.STL
> Parsing Mesh: <catkin_ws>src/cheetah/meshes/Hl1.STL
> Parsing Mesh: <catkin_ws>src/cheetah/meshes/Sl1.STL
> Parsing Mesh: <catkin_ws>src/cheetah/meshes/Body.STL
> Warning: This mesh has a lot of triangles!
> Warning: It is recommended to use the script with the '--disable-mesh-optimization' argument.
> Parsing Mesh: <catkin_ws>src/cheetah/meshes/FS.STL
> Warning: This mesh has a lot of triangles!
> Warning: It is recommended to use the script with the '--disable-mesh-optimization' argument.
> Parsing Mesh: <catkin_ws>src/cheetah/meshes/L1.STL
> Parsing Mesh: <catkin_ws>src/cheetah/meshes/L2.STL
> Parsing Mesh: <catkin_ws>src/cheetah/meshes/L3.STL
> Parsing Mesh: <catkin_ws>src/cheetah/meshes/Hb.STL
> Parsing Mesh: <catkin_ws>src/cheetah/meshes/Vb.STL
> Parsing Mesh: <catkin_ws>src/cheetah/meshes/FSb.STL
> Parsing Mesh: <catkin_ws>src/cheetah/meshes/RS.STL
> Warning: This mesh has a lot of triangles!
> Warning: It is recommended to use the script with the '--disable-mesh-optimization' argument.
> Parsing Mesh: <catkin_ws>src/cheetah/meshes/RSb.STL
> Parsing Mesh: <catkin_ws>src/cheetah/meshes/L4.STL
> Root link: world
> There are 52 links, 51 joints and 2 sensors
> ```
> 
> and the error still is there😰 , some `boundingObject`s still are `Sphere`s
> hare are screenshots of the generated proto file:
> [image]
> [image]
> [image]
> 
> The good news is that inertiaMatrix is being generated, though I don't know if it's generating the right results

I'm so sorry that I checked my urdf file and I find that I use different geometries in the `<collision>` and `<visual>` label. The master branch of the tool is working correctly.

closed 2020-09-18 by zapplelove

DavidMansolino, 2020-09-18:
> I'm so sorry that I checked my urdf file and I find that I use different geometries in the <collision> and <visual> label. The master branch of the tool is working correctly.

That makes way more sense, thank you for the feedback!