cyberbotics/urdf2webots #88: Fixed joints seem to be ignored during conversion

AndrewJSchoen, 2020-09-20:
Hi all,

I was using this tool with a couple different robots (franka emika panda and ur3e+robotiq_85) and it seems that the presence of fixed joints seem to be ignored. That is, when opened in Webots, any part of the robot after that fixed joint disconnects and falls due to gravity. As a test, I converted those joints to revolute ones, making them essentially fixed through the limits, and this seemed to fix the issue. I haven't fully tested the behavior of the robots with a controller, but will be doing so soon.
Attached is the panda urdf file that had the issues.

[panda.txt](https://github.com/cyberbotics/urdf2webots/files/5251924/panda.txt)

astumpf, 2020-09-24:
I can confirm this behavior. Currently our robots just falls apart, when using the current version of the exporter. Looks funny but makes the exporter currently unusable. I will try an older version from the repository.

Best regards

astumpf, 2020-09-24:
> Hi all,
>
> I was using this tool with a couple different robots (franka emika panda and ur3e+robotiq_85) and it seems that the presence of fixed joints seem to be ignored. That is, when opened in [...]

In my case, following commit is the issue: https://github.com/cyberbotics/urdf2webots/commit/8e5b0e244934c15b7fe90a503d29bcc17ec218b0

Maybe your links are missing a mass?

However, some robot URDF contain "fake" links. Especially the "base_link" can be a such a virtual link for representation reasons. In my case the base_link was indeed without a mass, resulting in disconnected robot parts. The linked patch introduces this behavior.

AndrewJSchoen, 2020-09-24:
In the case of the panda example, all the links had no mass. I also did this for the UR5+robotiq 85 gripper, and while a number of those links have mass, the link connecting the robot to the gripper did not.

pnp-erion, 2021-07-09:
I encountered this bug recently, too. Its sad that this issue was not yet solved despite being open for over nine months. After some time I found the root cause. Its not the fixed joints or missing mass but the dummy links having no collision.

For everyone encountering this bug in the future: Just add a <collision> object e.g. a box with sidelength 0.0 to all empty links in your URDF/xacro (like base_link).  That should fix the problem consistently (changing fixed joints to "revolute" didn't work for me). My hypothesis is that Webots requires a bounding object to simulate the physics of a solid and its connection and if there is no bounding object the physics calculation will fail.

Would be nice if this was fixed permanently because many URDFs contain dummy links and changing them is maybe not hard, but still annoying.

BenjaminHug8, 2022-02-04:
I am closing this as it has been solved in #146 by removing all the links not specifying a mass, see [here](https://github.com/cyberbotics/urdf2webots/blob/f0a990cccdc28641db3c38639fd6f5d7a8e150fd/urdf2webots/parserURDF.py#L856-L896).

closed 2022-02-04 by BenjaminHug8