cyberbotics/urdf2webots #89: Add option to refer to mesh files instead of deep copying meshes into proto files

astumpf, 2020-09-26:
Hi,

Currently the urdf exporter deep copies all meshes from any sources. This potentially bloats the proto files to multiple hundreds of megabytes, makes them not well editable anymore.

It would be very convenient to have an option that the exporter refers to files instead of deep copying their content. I've already implemented a kind of prototype for my case. I would appreciate seeing such a feature officially supported in the future. Feel free to use my code as a starting point: https://github.com/astumpf/urdf2webots/commit/07348925afcf134bb7371a522b33d2e072168c0c

Best regards

omichel, 2020-09-28:
That seems to be a very good idea and your implementation seems to be a good starting point.
I have however some concerns with the `--mesh-transform` option. Do we really need it? To me it seems that the same transform should always apply, so it could be hard-coded, isn't it?
Anyway, could you please open a Pull Request on this github repo so that we can review your contribution and eventually merge it?

astumpf, 2020-10-01:
> That seems to be a very good idea and your implementation seems to be a good starting point.
Thanks! :-)

> I have however some concerns with the `--mesh-transform` option. Do we really need it? To me it seems that the same transform should always apply, so it could be hard-coded, isn't it?
Yes, that's something I would expect too. However, when the mesh points are exported to the protos files, everything is correctly orientated. But, as soon you just replace these mesh points with a reference to an external collada file, the orientation is wrong. Seems there is going something wrong, when Webots is loading collada files. Therefore, I've added this option as workaround.

> Anyway, could you please open a Pull Request on this github repo so that we can review your contribution and eventually merge it?
Done: https://github.com/cyberbotics/urdf2webots/pull/90

BenjaminHug8, 2022-02-04:
I am closing this issue as the mesh files are not anymore converted by the `urdf2webots` tool but rather by the Assimp library when Webots parse worlds and protos.

closed 2022-02-04 by BenjaminHug8