cyberbotics/urdf2webots #92: Center of Mass not exported

martimorta, 2020-11-22:
Hi,
I have tried the converter with a simple differential drive robot [dd3.urdf](https://github.com/cyberbotics/urdf2webots/files/5579394/dd3.log) and it seems it doesn't add the center of mass giving this output

```
    INFO: dd3 (PROTO) > Solid > Physics: You must also specify the 'centerOfMass' when specifying the 'inertiaMatrix'
    WARNING: dd3 (PROTO) > Solid > Physics: 'centerOfmass' must also be specified when using an inertia matrix.
```
I tried adding an explicit origin tag in the URDF but that didn't help.

Might be off-topic: Adding the centerOfMass manually to each Solid, when I try to visualise it in Webots it shows this error which I am not sure it is related to the above
```
 INFO: DEF dd3 dd3 (PROTO): A Solid with a non-NULL Physics node must be chosen.
```
Thanks!

omichel, 2020-11-23:
Thank you for this bug report.
The warning you get is simply because you are trying to visualize the center of mass of a Solid node which has no Physics (it's physics field is set to NULL) and therefore, it has no center of mass. Hence Webots cannot visualize any center of mass for this Solid node.

omichel, 2020-11-23:
Could you test the version proposed in #93 and let me know if that fixes the problem?

martimorta, 2020-12-16:
Hi Olivier, I can confirm #93 fixes the problem. Thanks for the explanation about the physics, just trying Webots for the first time!

PR #93 (2021-02-04): Fix export of centerOfMass
Fixes #92.
The default value of the center of mass is not (0, 0, 0), but an empty array.
Therefore we should export the value of the center of mass if it is (0, 0, 0).
closed 2020-12-16 by martimorta