cyberbotics/urdf2webots #99: Camera orientation

jgueldenstein, 2021-01-15:
In a URDF the camera optical frame is usually specified as per https://www.ros.org/reps/rep-0103.html#suffix-frames

Webots uses a different standard described here https://www.cyberbotics.com/doc/reference/viewpoint

Therefore the camera is looking in the wrong direction and it is upside down as seen here
[image]

The problem to the solution can be to introduce an additional Transform in the export function of the Camera class https://github.com/cyberbotics/urdf2webots/blob/db4be076c2f867f4cad0985ec9d13074b04f4bdb/urdf2webots/parserURDF.py#L288
such that the camera is rotated in relation the the camera optical frame.

```
Solid {
  translation 0.043200 0.080000 -0.023500
  rotation 0.707105 -0.000001 0.707108 3.141591
  children [
    Transform {
    rotation 1.0 0.0 0.0 3.141591
    children [
      Camera {
        name "camera"
        fieldOfView 0.570000
        width 620
        height 480
        near 0.08
      }
      ]
    }
  ]
  name "camera_optical_frame"
}
```
I can provide a PR if you are interested.

omichel, 2021-01-15:
> I can provide a PR if you are interested.

Yes please.

Note: it may be simpler to set the rotation inside the `Camera` node itself:
```js
Solid {
  translation 0.043200 0.080000 -0.023500
  rotation 0.707105 -0.000001 0.707108 3.141591
  children [
    Camera {
      rotation 1.0 0.0 0.0 3.141591
      name "camera"
      fieldOfView 0.570000
      width 620
      height 480
      near 0.08
    }
  ]
  name "camera_optical_frame"
}
```

jgueldenstein, 2021-01-15:
Thanks for the suggestion, I did not now a camera could have a rotation member, but it makes sense since it's derived from solid which is derived from transform.
https://github.com/cyberbotics/urdf2webots/pull/100 is the PR.

omichel, 2021-01-15:
Fixed in #100.

closed 2021-01-15 by omichel