cyberbotics/urdf2webots #103: Treat collada.material.Effect when returns collada.material.Map rather than Tuple

jgvictores, 2021-01-16:
Hello!

I've been using `urdf2webots` to port some humanoid robots to Webots (e.g. https://github.com/roboticslab-uc3m/teo-webots-models lacks textures and dynamics but serves its purpose). While porting PAL Robotics REEM-C robot, I ran into `TypeError: 'Map' object does not support indexing` upon passing through `visual.material.specular = colorVector2Instance(data.material.effect.specular)`. Here are the robot model repositories:
- https://github.com/pal-robotics/reemc_simulation
  - https://github.com/pal-robotics/reemc_robot

It turns out a Python `tuple`is expected; however, https://pycollada.readthedocs.io/en/latest/reference/generated/collada.material.Effect.html#collada.material.Effect as documented can **alternatively return** an instance of [collada.material.Map](https://pycollada.readthedocs.io/en/latest/reference/generated/collada.material.Map.html#collada.material.Map), which is the case for this robot. The shameless (and textureless) hack at https://github.com/cyberbotics/urdf2webots/blob/master/urdf2webots/parserURDF.py#L553 ([perma](https://github.com/cyberbotics/urdf2webots/blob/aa6706adfc5c8039089aca0a330b54036d5386d4/urdf2webots/parserURDF.py#L553)) was the following:

```diff
-                    if data.material.effect.specular:
+                    if data.material.effect.specular and isinstance(data.material.effect.specular, tuple):
```

I understand the expected behavior would be to treat the different situations where https://pycollada.readthedocs.io/en/latest/reference/generated/collada.material.Effect.html#collada.material.Effect returns a `collada.material.Map` (which is not restricted to `specular`), and/or at least prevent the `TypeError`. Thank you!

omichel, 2021-01-18:
I believe we could implement your patch until we support the `collada.material.Map` type. Could you open a PR for that?

jgvictores, 2021-01-18:
Sure! Here it goes: https://github.com/cyberbotics/urdf2webots/pull/105

omichel, 2021-01-19:
Fixed in #105.

closed 2021-01-19 by omichel