cyberbotics/urdf2webots #109: How to deal with controllers and sensors

reinzor, 2021-03-01:
Hi,

We're totally new to `webots` (we have been using `gazebo` so far) and we were wondering how to specify sensors and other plugins, e.g. controllers in the urdf. A lidar sensor in URDF (for gazebo) would be something like:

```
<gazebo reference="front_scan">
    <sensor name="front_scan" type="gpu_ray">
      <update_rate>34</update_rate>
      <pose>0 0 0 0 0 0</pose>
      <visualize>false</visualize>
      <ray>
        <scan>
          <horizontal>
            <samples>1072</samples>
            <resolution>1</resolution>
            <min_angle>-1.5533430342749535</min_angle>
            <max_angle>1.5533430342749535</max_angle>
          </horizontal>
        </scan>
        <range>
          <min>0.08</min>
          <max>30.0</max>
          <resolution>0.01</resolution>
        </range>
        <noise>
          <type>gaussian</type>
          <mean>0.0</mean>
          <stddev>0.001</stddev>
        </noise>
      </ray>
      <plugin filename="libgazebo_ros_gpu_laser.so" name="gazebo_ros_front_scan_controller">
        <topicName>front_scan</topicName>
        <frameName>front_scan</frameName>
      </plugin>
    </sensor>
  </gazebo>
```

How does this work for `webots`. We have tried the exporter and all links get exported (including geometry) but how to specificy sensors / actuators / plugins? We would like to maintain one robot description (like we have now).

Thanks!

jgueldenstein, 2021-03-01:
The issue is, that urdf2webots as of now only parses lidar sensors of `type="ray"`

I made a PR #111 to fix this.

closed 2021-03-02 by omichel

reinzor, 2021-03-02:
@jgueldenstein, so this parser parses the gazebo tags and creates a proto model for this with the related sensors. But how do we point to custom plugins that we currently have. According to your documentation, this should be handled in the robot `controller`. How can we point to this controller and can we pass parameters via the urdf file to the controller?

jgueldenstein, 2021-03-02:
@reinzor In webots controllers work a bit different than sensor plugins in gazebo.
In webots you define your sensors and their properties such as field of view or resolution in the `.proto` file.
There are some existing controllers such as the webots_ros controller provide a service based interface to all the devices on your robot. You can also choose to write your own controller to keep the interface that you are using with the rest of your software or write a middle man software that translates between services and ros topics.

reinzor, 2021-03-03:
Thank @jgueldenstein , I understand this from your documentation. However, is it also possible using this conversion tool to keep all configuration in the `urdf`? So specify the used controller (that is being written to the `.proto` file) and its arguments in the `urdf`? Otherwise, a manual step would always be required for adding the `controller` and `arguments` to the `.proto` file. So we need to have both the `urdf` and `proto` in version management.

jgueldenstein, 2021-03-03:
@reinzor the controller that is being loaded is defined by the controller field in the Robot node of the `.proto` file. It is exported such that it can be changed in the webots simulation interface and saved in a webots world file. This way you do not need the `.proto` file but only the world file in version management.

jgueldenstein, 2021-03-03:
Could you specify what you mean by `arguments`?

reinzor, 2021-03-03:
I mean the controller arguments as mentioned here https://cyberbotics.com/doc/guide/controller-programming#using-controller-arguments .

> It is exported such that it can be changed in the webots simulation interface and saved in a webots world file. 

Thanks, this clears things up! Thanks for all your quick replies!