cyberbotics/urdf2webots #120: sensorElement always refers to the first element of a list

ManuelZ, 2021-06-01:
I think [this line](
https://github.com/cyberbotics/urdf2webots/blob/ce63025d17c2343eef47e1b523a38ce3edaecae0/urdf2webots/parserURDF.py#L948) shouldn't exist because in the previous line, where you are looping, you already have a `sensorElement`.

I have 4 `<sensor>` tags with different types of cameras and I'm getting 4 cameras of the same first type.
```
for sensorElement in element.getElementsByTagName('sensor'):
    sensorElement = element.getElementsByTagName('sensor')[0]
```

omichel, 2021-06-02:
I agree. I sent you an invite to become a member of the Webots Committers group. May I ask you to open a PR for removing this line?

ManuelZ, 2021-06-02:
Alright, just did.

https://github.com/cyberbotics/urdf2webots/pull/121

closed 2021-06-02 by ManuelZ