cyberbotics/urdf2webots #123: Need extensionSlot in generated proto file

ManuelZ, 2021-06-21:
I'm trying to recreate the [webots_ros2_turtlebot](https://github.com/cyberbotics/webots_ros2/tree/master/webots_ros2_turtlebot) example with a custom robot and use it with the new plugin system. I [was](https://discord.com/channels/565154702715518986/565154703139405824/856372217780502528) facing some trouble with an error message like this:

```
[driver-3] Warning: "inertial_unit" device not found.
[driver-3] terminate called after throwing an instance of 'std::runtime_error'
[driver-3]   what():  Cannot find InertialUnit with name inertial_unit
```
I found out that my generated PROTO file, based on an URDF, is missing something like this for my world's `extensionSlot` to work:
```
field MFNode     extensionSlot   [ ]
```

A line like this should probably go [here](https://github.com/cyberbotics/urdf2webots/blob/6630d9778af064983f97ef1b2ea87f91c1efb48b/urdf2webots/writeProto.py#L71):
```
proto.write('  field  MFNode      extensionSlot   []    ' + spaces + '# Extends the robot with new nodes in the extension slot.\n')
```

But something like this seems to be needed as well (based on [this](https://raw.githubusercontent.com/cyberbotics/webots_ros2/master/webots_ros2_turtlebot/protos/TurtleBot3Burger_enu.proto)):
```
    Transform {
        rotation 0.5773503 -0.5773503 -0.5773503 2.0943951
        translation -0.03 0 0.153
        children IS extensionSlot
      }
```

But I'm not sure how to add that in a generic way.

lukicdarkoo, 2021-06-21:
There is a `--tool-slot=linkName` argument. Does it solve your problem?

ManuelZ, 2021-06-21:
Not really. I did add `--tool-slot="imu_link"` to my `ros2 run webots_ros2_importer xacro2proto` call, but the output after running the robot launch file was the same.

I realized that I could as well point out to my already available generated IMU devices, so I did, and that solved my current issues. But mentioning this extension slot generation thing seemed like a good idea, for completeness.

Besides that, I'm using a mobile robot. The `toolSlot` docs that I've found mention: 
```toolSlot: Extend the robot with new nodes at the end of the arm.```

And the `extensionSlot` doc says:
```extensionSlot: Extension slot allowing the user to add other nodes (e.g., sensors, shape of the vehicle, etc.).```

I'm not really sure if they actually fulfill the same purpose, but at least in the docs they are being used in different robots contexts.  Sounds like if the `toolSlot` should be used specifically in robot arms. 

And as well, the `--tool-slot` flag talks about a URDF link name and the extension slot is used with a PROTO node.

lukicdarkoo, 2021-06-22:
> Besides that, I'm using a mobile robot. The `toolSlot` docs that I've found mention:
> `toolSlot: Extend the robot with new nodes at the end of the arm.`
>
> And the `extensionSlot` doc says: [...]

It is only a difference in a naming convention, both are Slot nodes. However, you are right, a more generic solution would be to an argument something like:
```
--slot=slot_link,slot_name
```

> Not really. I did add `--tool-slot="imu_link"` to my `ros2 run webots_ros2_importer xacro2proto` call, but the output after running the robot launch file was the same.

Does the `imu_link` exist in your URDF? You should also be able to `--tool-slot="base_link"` if the IMU is not attached to an arm or similar.

ManuelZ, 2021-06-25:
Sorry, I was making a mistake with the names here:
```
    <inertialUnitName>imu inertial</inertialUnitName>
    <gyroName>imu gyro</gyroName>
    <accelerometerName>imu accelerometer</accelerometerName>
```
I can rely now on using `--tool-slot`. No need for `extensionSlot`. Thanks.

By the way, for whoever gets here, `--tool-slot` adds a physics node to the link that you refer to:
https://github.com/cyberbotics/urdf2webots/blob/6630d9778af064983f97ef1b2ea87f91c1efb48b/urdf2webots/writeProto.py#L142
And in my case, where I'm referring to a link for an imu, that is needed in the parent.

closed 2022-01-14 by BenjaminHug8