cyberbotics/urdf2webots #131: a bug about miss joint label

chinahuangyong, 2021-09-29:
[Stdbot.zip](https://github.com/cyberbotics/urdf2webots/files/7250649/Stdbot.zi
[stdbot.zip](https://github.com/cyberbotics/urdf2webots/files/7250650/stdbot.zip)
p)

miss a HingeJoint tag at the line 359 int the .proto file.

[image]

BenjaminHug8, 2022-03-10:
Can you say which `HingeJoint` is missing?

(that would mean that a joint of type `revolute` is not exported, I have check the PROTO that I have converted now and all the joints of type `revolute` have a corresponding `HingeJoint`)

chinahuangyong, 2022-03-10:
i have 4 `joint` link to body in urdf file, but it get only three HingeJoint in proto file. 

joint LF to body in urdf file Line 88:

```
  <joint
    name="TV_LF_Joint1"
    type="revolute">
    <origin
      xyz="0.22 0.0516 -0.072"
      rpy="0 0 0" />
    <parent
      link="base_link" />
    <child
      link="TV_LF_Link1" />
    <axis
      xyz="0 1 0" />
    <limit
      lower="0"
      upper="0"
      effort="0"
      velocity="0" />
  </joint>
```

joint RF to body in urdf file line 373:
```
<joint
    name="TV_RF_Joint1"
    type="fixed">
    <origin
      xyz="0.22 -0.0516 -0.072"
      rpy="0 0 0" />
    <parent
      link="base_link" />
    <child
      link="TV_RF_Link1" />
    <axis
      xyz="0 0 0" />
    <limit
      lower="0"
      upper="0"
      effort="0"
      velocity="0" />
  </joint>
```
and othes at Line 658 and Line 943 in urdf file.

BenjaminHug8, 2022-03-10:
> i have 4 `joint`s link to body in urdf file, but it get only three `HingeJoint`s in proto file.

That is not a surprise, you have a `fixed` type joint

> joint RF to body in urdf file line 373:
>
> ```
> <joint
>     name="TV_RF_Joint1"
>     type="fixed">
>     <origin
>       xyz="0.22 -0.0516 -0.072"
>       rpy="0 0 0" />
>     <parent
>       link="base_link" [...]

`Fixed` type joints are not converted into `HingeJoint`, but rather the `rpy` and `xyz` of the joint is used for the translation and rotation of the child link that is converted into a `Solid` node (the one in line 359 on the image you posted).

closed 2022-03-10 by BenjaminHug8