cyberbotics/urdf2webots #145: error: Did not find a 'DEF base_link' to match 'USE base_link'.

vector-you, 2022-01-11:
[wgrobot.zip](https://github.com/cyberbotics/urdf2webots/files/7847578/wgrobot.zip)

BenjaminHug8, 2022-01-11:
It seems that the conversion from URDF to PROTO works but some joints are try to `USE` without a corresponding `DEF`.

Check if something is not reset or faking an already converted joint.

BenjaminHug8, 2022-01-14:
I do not get the same errors than you (I am on Ubuntu).

In the PROTO file, I get things like:

```
geometry DEF C:\Users\vecto\Desktop\wgrobot\meshes\cylinder1 Mesh {
  url "/home/benjamin/catkin_ws/src/wgrobot/urdf/C:\Users\vecto\Desktop\wgrobot\meshes\cylinder1.STL"
}
```

For example changing `mesh
          filename="C:\Users\vecto\Desktop\wgrobot\meshes\base_link.STL" />` by
`<mesh filename="../meshes/base_link.STL" />` in your URDF file gives me a correct output like this in the PROTO file:

```
geometry DEF base_link Mesh {
  url "/home/benjamin/catkin_ws/src/wgrobot/meshes/base_link.STL"
}
```

I will check if I get the same thing under Windows.

vector-you, 2022-01-14:
I just followed your example by changing the mesh filename path format, and I got the PROTO file whose output like this :
geometry DEF base_link Mesh {
          url "C:\Users\vecto\Desktop/tail_simple/meshes/base_link.STL"
        }
I loaded the PROTO file in the webots , but the same problem also occured.
Then I changed the PROTO file like this:
geometry DEF base_link Mesh {
          url "C:/Users/vecto/Desktop/tail_simple/meshes/base_link.STL"
        }
I can get the correct model in the webots.
Now it works.
Thank you very much for your help,Benjamin Hug.

BenjaminHug8, 2022-01-14:
> I will check if I get the same thing under Windows.

I got a correct output on Windows without changing the URDF file. I downloaded the 1.0.13 version of `urdf2webots` and it worked fine.

Changing manually your PROTO file is a way to solve this but you should be able to make the conversion tool work too. Anyway I am happy you managed to make your PROTO file working!

closed 2022-01-14 by BenjaminHug8