cyberbotics/urdf2webots #152: How to reflect meshes loaded from files?

astumpf, 2022-02-04:
Robots are usually symmetrical and therefore many parts are identical but reflected. Using URDF it's common practice to just use the identical mesh multiple times and just reflect it by using negative scales, e.g. for mirroring along the x-z plane (sagittal) you would use the scaling vector [1 -1 1]. However, Webots forbids this type of mesh reflection using the Transform node (https://cyberbotics.com/doc/reference/transform).

```
Transform {
  translation 0.000000 0.000000 0.060000
  rotation 0.000000 0.000000 1.000000 0.000000
  scale 1.000000 -1.000000 1.000000
  children [
    Shape {
      appearance PBRAppearance {
        baseColor 0.500000 0.500000 0.500000
        transparency 0.000000
        roughness 1.000000
        metalness 0
        emissiveColor 0.000000 0.000000 0.000000
      }
      geometry DEF my_mesh Mesh {
        url "my_awesome_mesh.obj"
      }
    }
  ]
  }
```

Is there any way to reuse a single mesh file in case of reflections?

BenjaminHug8, 2022-02-07:
Hi @astumpf,

As you have maybe noticed from the doc you mentioned, scale cannot be negative in Webots.

However I think we should be able to add boolean in the `Mesh` node to mirrors the mesh along a certain plane. I just need to look how URDF specifies this mirrors effects to know how we could use it in the conversion tool.

Thank you for the proposition!

astumpf, 2022-02-07:
Sounds good! I am already looking forward to the update.

BenjaminHug8, 2022-02-25:
@astumpf the PR #157 has been merged in the `develop` branch and should solve this issue. Feel free to reopen if needed.

closed 2022-02-25 by BenjaminHug8