cyberbotics/urdf2webots #153: Single appearance feature for meshfiles overwrites transforms

astumpf, 2022-02-04:
In example:
https://github.com/cyberbotics/urdf2webots/blob/f0a990cccdc28641db3c38639fd6f5d7a8e150fd/urdf2webots/parserURDF.py#L680

This causes the visual to lose its individual properties such as transformation. The assumption that the same meshfile will be re-used with the identical transformation is wrong.

BenjaminHug8, 2022-02-07:
Thank you for the report, I will take a look.

BenjaminHug8, 2022-02-07:
I just tested and everything is fine on my side, I can use two links with the same mesh and have to joints with different origins and the results is ok.

The visual object in our tool contains a translation and rotation:
https://github.com/cyberbotics/urdf2webots/blob/f0a990cccdc28641db3c38639fd6f5d7a8e150fd/urdf2webots/parserURDF.py#L159-L167
and as you can see the geometry is variable of this class.

The line you mentioned is just for reusing the same `Mesh` node contained in the geometry, but that's all.
Let me know if you still have questions or if your issue is solved or not.

astumpf, 2022-02-07:
Yes, you're right. I overlooked this detail.  To give full detail about my concern: I'm working on a patch to consider also scaling of meshes. But the scale is indeed part of the `Geometry` object (https://github.com/cyberbotics/urdf2webots/blob/f0a990cccdc28641db3c38639fd6f5d7a8e150fd/urdf2webots/parserURDF.py#L159-L167) that will be overwritten as stated above.

BenjaminHug8, 2022-02-07:
> However, this scale patch would become obsolete in my case, when #152 was solved.

Yes exactly, I am implementing the `scale` option of `mesh` tags for solving this issue.

BenjaminHug8, 2022-02-25:
@astumpf the PR #157 has been merged and should solve your issue. Feel free to reopen this issue if needed.

closed 2022-02-25 by BenjaminHug8

astumpf, 2022-02-25:
Thanks a lot!