cyberbotics/urdf2webots #154: Webots does neither use color nor textures from wavefront (.obj) files

astumpf, 2022-02-04:
RViz can take use of color and texture information from the wavefront (.obj) files...

[image]

...while Webots cannot...

[image]

In both cases the same urdf with the identical file system structure is used. Any suggestions how to fix this issue?

BenjaminHug8, 2022-02-07:
Hello @astumpf, thanks for the report.

May I ask you your files so I can debug and track where is the issues in our code?

astumpf, 2022-02-07:
> Hello @astumpf, thanks for the report.
> 
> May I ask you your files so I can debug and track where is the issues in our code?

I can try to compile an example. Do you have any public available business mail I can send to?

omichel, 2022-02-07:
Yes, please send to support@cyberbotics.com.

BenjaminHug8, 2022-03-07:
@astumpf did you send a minimal example to support@cyberbotics.com so we can investigate this issue?
(Maybe you did but I can not found a mail related to this issue)

astumpf, 2022-04-05:
I've just sent a minimal example to your support mail. Please let me know if this works for you.

ad-daniel, 2022-04-07:
I think this is currently a limitation. URDF geometry tags (with obj files) are converted to Webots Mesh nodes, which as the name implies discards both textures and materials that might be included in the .obj file and merges all meshes into one.
On the Webots side we are in the process of merging a new node called `ColladaShape` https://github.com/cyberbotics/webots/pull/4439 which, as the name entails, provided a `.dae` file it generates both the mesh and the corresponding appearance.
In practice however, despite the name, I don't think there's fundamentally much difference in the handling of collada and wavefront files, so with limited changes to that PR it could (in theory) be made to support both.
It would however require a new name, to take inspiration from URDF it could be called `VisualShape` or something, it's the best I can come up with at the moment.
More importantly than the name, @omichel should we consider making a more generic node that supports both formats?

astumpf, 2022-04-07:
> I think this is currently a limitation. URDF geometry tags (with obj files) are converted to Webots Mesh nodes, which as the name implies discards both textures and materials that might be included in the .obj file and merges all meshes into one.

I'm currently not on my dev OS to 100% confirm that, but I'm pretty sure that currently the generated proto refers to the .obj file directly. The mesh is not included as Mesh nodes in the final proto. At least the provided minimal example does not result having any mesh nodes.

In the past, we've used collada files where Webots used to consider only color data but no textures. Unfortunately, I can't test this anymore.

ad-daniel, 2022-04-07:
> I'm currently not on my dev OS to 100% confirm that, but I'm pretty sure that currently the generated proto refers to the .obj file directly. The mesh is not included as Mesh nodes in the final proto. At least the provided minimal example does not result having any mesh nodes.

Yes, the Webots Mesh node is provided the url/path to the obj directly, but under the hood when Webots parses it, it strips all textures/materials.

> In the past, we've used collada files where Webots used to consider only color data but no textures. Unfortunately, I can't test this anymore.

That's how it used to behave, currently in the develop branch it generates a ColladaShapes node (technically it's a PROTO, which is what the new node will replace), so it includes both mesh and appearance for collada files.

omichel, 2022-04-07:
> I think this is currently a limitation. URDF geometry tags (with obj files) are converted to Webots Mesh nodes, which as the name implies discards both textures and materials that might be included in the .obj file and merges all meshes into one. On the Webots side we are in the process of merging a new node called `ColladaShape` [cyberbotics/webots#4285](https://github.com/cyberbotics/webots/pull/4285) which, as the name entails, provided a `.dae` file it generates both the mesh and the corresponding appearance. In practice however, despite the name, I don't think there's fundamentally much difference in the handling of collada and wavefront files, so with limited changes to that PR it could (in theory) be made to support both. It would however require a new name, to take inspiration from URDF it could be called `VisualShape` or something, it's the best I can come up with at the moment. More importantly than the name, @omichel should we consider making a more generic node that supports both formats?

Yes, that sounds like a very good idea.

astumpf, 2022-08-08:
I'm wondering when will there be an update in this regard, since a lot of progress was apparently already made a long time ago?

omichel, 2022-08-08:
Yes, we actually implemented the [CadShape](https://www.cyberbotics.com/doc/reference/cadshape?version=R2022b) node in Webots R2022b to be released very soon. That should resolve the issue.

astumpf, 2022-10-06:
I can confirm that this looks pretty nice now. Thanks a lot!