cyberbotics/urdf2webots #162: NameError: global name 'PackageNotFoundError' is not defined

DrakeProvost, 2022-02-22:
### System information
* Ubuntu 20.04.4 LTS
* Python 2.7.18
* Python 3.8.10

### Installation of `urdf2webots`
```
$ curl https://bootstrap.pypa.io/pip/2.7/get-pip.py | python
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100 1863k  100 1863k    0     0  9411k      0 --:--:-- --:--:-- --:--:-- 9411k
DEPRECATION: Python 2.7 reached the end of its life on January 1st, 2020. Please upgrade your Python as Python 2.7 is no longer maintained. pip 21.0 will drop support for Python 2.7 in January 2021. More details about Python 2 support in pip can be found at https://pip.pypa.io/en/latest/development/release-process/#python-2-support pip 21.0 will remove support for this functionality.
Defaulting to user installation because normal site-packages is not writeable
Collecting pip<21.0
  Downloading pip-20.3.4-py2.py3-none-any.whl (1.5 MB)
     |████████████████████████████████| 1.5 MB 11.3 MB/s 
Collecting setuptools<45
  Downloading setuptools-44.1.1-py2.py3-none-any.whl (583 kB)
     |████████████████████████████████| 583 kB 11.0 MB/s 
Collecting wheel
  Downloading wheel-0.37.1-py2.py3-none-any.whl (35 kB)
Installing collected packages: pip, setuptools, wheel
  WARNING: The scripts pip, pip2 and pip2.7 are installed in '/home/drakeprovost/.local/bin' which is not on PATH.
  Consider adding this directory to PATH or, if you prefer to suppress this warning, use --no-warn-script-location.
  WARNING: The scripts easy_install and easy_install-2.7 are installed in '/home/drakeprovost/.local/bin' which is not on PATH.
  Consider adding this directory to PATH or, if you prefer to suppress this warning, use --no-warn-script-location.
  WARNING: The script wheel is installed in '/home/drakeprovost/.local/bin' which is not on PATH.
  Consider adding this directory to PATH or, if you prefer to suppress this warning, use --no-warn-script-location.
ERROR: pip's legacy dependency resolver does not consider dependency conflicts when selecting packages. This behaviour is the source of the following dependency conflicts.
launch-ros 0.11.6 requires pyyaml, which is not installed.
Successfully installed pip-20.3.4 setuptools-44.1.1 wheel-0.37.1
```

```
$ python -m pip install urdf2webots
DEPRECATION: Python 2.7 reached the end of its life on January 1st, 2020. Please upgrade your Python as Python 2.7 is no longer maintained. pip 21.0 will drop support for Python 2.7 in January 2021. More details about Python 2 support in pip can be found at https://pip.pypa.io/en/latest/development/release-process/#python-2-support pip 21.0 will remove support for this functionality.
Defaulting to user installation because normal site-packages is not writeable
Collecting urdf2webots
  Downloading urdf2webots-1.0.16.tar.gz (30 kB)
Collecting pycollada>=0.6
  Using cached pycollada-0.7.2.tar.gz (107 kB)
Collecting Pillow
  Downloading Pillow-6.2.2-cp27-cp27mu-manylinux1_x86_64.whl (2.1 MB)
     |████████████████████████████████| 2.1 MB 11.0 MB/s 
Collecting numpy
  Downloading numpy-1.16.6-cp27-cp27mu-manylinux1_x86_64.whl (17.0 MB)
     |████████████████████████████████| 17.0 MB 10.9 MB/s 
Collecting rospkg
  Downloading rospkg-1.3.0.tar.gz (41 kB)
     |████████████████████████████████| 41 kB 205 kB/s 
Collecting python-dateutil>=2.2
  Downloading python_dateutil-2.8.2-py2.py3-none-any.whl (247 kB)
     |████████████████████████████████| 247 kB 10.8 MB/s 
Collecting catkin_pkg
  Downloading catkin_pkg-0.4.24.tar.gz (61 kB)
     |████████████████████████████████| 61 kB 7.9 MB/s 
Collecting PyYAML
  Downloading PyYAML-5.4.1-cp27-cp27mu-manylinux1_x86_64.whl (574 kB)
     |████████████████████████████████| 574 kB 10.8 MB/s 
Collecting distro
  Downloading distro-1.6.0-py2.py3-none-any.whl (19 kB)
Collecting six>=1.5
  Downloading six-1.16.0-py2.py3-none-any.whl (11 kB)
Collecting docutils
  Downloading docutils-0.18.1-py2.py3-none-any.whl (570 kB)
     |████████████████████████████████| 570 kB 10.8 MB/s 
Collecting pyparsing
  Downloading pyparsing-2.4.7-py2.py3-none-any.whl (67 kB)
     |████████████████████████████████| 67 kB 7.7 MB/s 
Building wheels for collected packages: urdf2webots, pycollada, rospkg, catkin-pkg
  Building wheel for urdf2webots (setup.py) ... done
  Created wheel for urdf2webots: filename=urdf2webots-1.0.16-py2-none-any.whl size=29277 sha256=d8b082c761b2ee8c25eb3f31f0caee24c18753997950256edbafa8a62e0f0a1a
  Stored in directory: /home/drakeprovost/.cache/pip/wheels/93/58/4d/9d09a6d750bd527560cfd4aee5688d64aab36152450d7ec329
  Building wheel for pycollada (setup.py) ... done
  Created wheel for pycollada: filename=pycollada-0.7.2-py2-none-any.whl size=127010 sha256=b87b874f71c08e61c3e69d0d8d485308ce378900e2050e220cc0dd5367c9a0bd
  Stored in directory: /home/drakeprovost/.cache/pip/wheels/45/ae/8a/01f8ebbc2b52f5ab297fd932ddb9228b84e80a0bc7c851efc3
  Building wheel for rospkg (setup.py) ... done
  Created wheel for rospkg: filename=rospkg-1.3.0-py2-none-any.whl size=35571 sha256=1e5c51aed75d720d3c07ad45d3aad5e7fd543d19568fcb971d11e4c64bbb84ee
  Stored in directory: /home/drakeprovost/.cache/pip/wheels/5f/a3/72/d6457a71e4028d86a4c4a7edeef9de4cd275b4133374ebed3b
  Building wheel for catkin-pkg (setup.py) ... done
  Created wheel for catkin-pkg: filename=catkin_pkg-0.4.24-py2-none-any.whl size=74982 sha256=3ba7f5dfa10664fe0297d70f5b240411cb060ae97437fce128bb378c00627057
  Stored in directory: /home/drakeprovost/.cache/pip/wheels/53/1e/dc/ab9bc967b639a0c879ce348af6a8cede07fee48b666b2b8add
Successfully built urdf2webots pycollada rospkg catkin-pkg
Installing collected packages: six, python-dateutil, numpy, pycollada, Pillow, docutils, pyparsing, catkin-pkg, PyYAML, distro, rospkg, urdf2webots
  WARNING: The scripts f2py, f2py2 and f2py2.7 are installed in '/home/drakeprovost/.local/bin' which is not on PATH.
  Consider adding this directory to PATH or, if you prefer to suppress this warning, use --no-warn-script-location.
  WARNING: The scripts catkin_create_pkg, catkin_find_pkg, catkin_generate_changelog, catkin_package_version, catkin_prepare_release, catkin_tag_changelog and catkin_test_changelog are installed in '/home/drakeprovost/.local/bin' which is not on PATH.
  Consider adding this directory to PATH or, if you prefer to suppress this warning, use --no-warn-script-location.
  WARNING: The script distro is installed in '/home/drakeprovost/.local/bin' which is not on PATH.
  Consider adding this directory to PATH or, if you prefer to suppress this warning, use --no-warn-script-location.
  WARNING: The script rosversion is installed in '/home/drakeprovost/.local/bin' which is not on PATH.
  Consider adding this directory to PATH or, if you prefer to suppress this warning, use --no-warn-script-location.
Successfully installed Pillow-6.2.2 PyYAML-5.4.1 catkin-pkg-0.4.24 distro-1.6.0 docutils-0.18.1 numpy-1.16.6 pycollada-0.7.2 pyparsing-2.4.7 python-dateutil-2.8.2 rospkg-1.3.0 six-1.16.0 urdf2webots-1.0.16
```

### Expected Output
I used the command `python -m urdf2webots.importer --input=Rover_2022.urdf --output=Rover_2022.proto`, which should've generated the proto file.

### Broken Output
```
$ python -m urdf2webots.importer --input=Rover_2022.urdf --output=Rover_2022.proto
Traceback (most recent call last):
  File "/usr/lib/python2.7/runpy.py", line 174, in _run_module_as_main
    "__main__", fname, loader, pkg_name)
  File "/usr/lib/python2.7/runpy.py", line 72, in _run_code
    exec code in run_globals
  File "/home/drakeprovost/.local/lib/python2.7/site-packages/urdf2webots/importer.py", line 334, in <module>
    options.enableMultiFile, options.toolSlot, options.initTranslation, options.initRotation, options.initPos, options.linkToDef, options.jointToDef, options.relativePathPrefix)
  File "/home/drakeprovost/.local/lib/python2.7/site-packages/urdf2webots/importer.py", line 89, in convertUrdfFile
    initPos, linkToDef, jointToDef, relativePathPrefix)
  File "/home/drakeprovost/.local/lib/python2.7/site-packages/urdf2webots/importer.py", line 174, in convertUrdfContent
    except PackageNotFoundError:
NameError: global name 'PackageNotFoundError' is not defined
```

BenjaminHug8, 2022-02-22:
Thank you for raising this issue, I will fix that!

A (possible) temporary fix would be to source your ROS distribution, let me know if that works for you.

BenjaminHug8, 2022-02-22:
May I ask you your ROS distribution?

BenjaminHug8, 2022-02-22:
It seems like you are missing the `ros-DISTRO-ament-index-python`, where `DISTRO` is your ros distribution.

If it is already installed, can you show your `PYTHONPATH`?

DrakeProvost, 2022-02-22:
```
$ printenv | grep -i ROS
ROS_VERSION=2
ROS_PYTHON_VERSION=3
AMENT_PREFIX_PATH=/opt/ros/foxy
PYTHONPATH=/opt/ros/foxy/lib/python3.8/site-packages
_colcon_cd_root=/home/drakeprovost/ros2_install
LD_LIBRARY_PATH=/usr/lib/x86_64-linux-gnu/gazebo-11/plugins:/opt/ros/foxy/opt/yaml_cpp_vendor/lib:/opt/ros/foxy/opt/rviz_ogre_vendor/lib:/opt/ros/foxy/lib/x86_64-linux-gnu:/opt/ros/foxy/lib
ROS_LOCALHOST_ONLY=0
PATH=/opt/ros/foxy/bin:/sbin:/bin:/usr/bin:/usr/local/bin:/snap/bin
ROS_DISTRO=foxy
```

BenjaminHug8, 2022-02-22:
> ```
> $ printenv | grep -i ROS
> ROS_VERSION=2
> ROS_PYTHON_VERSION=3
> AMENT_PREFIX_PATH=/opt/ros/foxy
> PYTHONPATH=/opt/ros/foxy/lib/python3.8/site-packages
> _colcon_cd_root=/home/drakeprovost/ros2_install [...]

It seems all good to me, have you checked if `ros-foxy-ament-index-python` is installed?

Could you try to use python3 instead of python2 when running the script?

DrakeProvost, 2022-02-22:
Forgive my ignorance, can you tell me how to check if `ros-foxy-ament-index-python` is installed?

Trying python3 does seem to have accomplished something (if python2 doesn't work, it might be worth updating [the command in the README](https://github.com/cyberbotics/urdf2webots#from-pip-1)):
```
$ python3 -m urdf2webots.importer --input=Rover_2022.urdf --output=Rover_2022.proto
Package "Rover_2022" not found.
Can't determine package root path.
Package "Rover_2022" not found.
Can't determine package root path.
Package "Rover_2022" not found.
Can't determine package root path.
Package "Rover_2022" not found.
Can't determine package root path.
Package "Rover_2022" not found.
Can't determine package root path.
Package "Rover_2022" not found.
Can't determine package root path.
Package "Rover_2022" not found.
Can't determine package root path.
Package "Rover_2022" not found.
Can't determine package root path.
Package "Rover_2022" not found.
Can't determine package root path.
Package "Rover_2022" not found.
Can't determine package root path.
Root link: Chassis
There are 5 links, 4 joints and 0 sensors
$ ls
README.md  Rover_2022.csv  Rover_2022.proto  Rover_2022_textures  Rover_2022.urdf
```
(In response to the "package not found" stuff, I see I have in `Rover_2022.urdf` a lot of `<mesh filename="package://Rover_2022/meshes/Chassis.STL" />` but haven't put the meshes repo in my directory, so that one's on me. I'll update this comment once I've done that)

BenjaminHug8, 2022-02-23:
> Forgive my ignorance, can you tell me how to check if `ros-foxy-ament-index-python` is installed?

One way to do it is to run `dpkg -l ros-foxy-ament-index-python` in a terminal, but as python3 is working that means it is installed.

> Trying python3 does seem to have accomplished something

Nice :)

As ROS 2 is working with python3 you probably cannot run the conversion tool with python2.

> (if python2 doesn't work, it might be worth updating [the command in the README](https://github.com/cyberbotics/urdf2webots#from-pip-1)):

I guess it might help users you are right, but as you can use the tool with ROS 1 (which is suppose to work with python2 until `noetic` I guess) I will add something accordingly.

> ```
> $ python3 -m urdf2webots.importer --input=Rover_2022.urdf --output=Rover_2022.proto
> Package "Rover_2022" not found.
> Can't determine package root path.

This error mainly says that no  "Rover_2022" package has been found with ROS 2, you either need to install the package and then re source foxy, or if it is your package you should build it and then source it.
[Here](https://github.com/cyberbotics/webots_ros2/blob/develop/webots_ros2_universal_robot/resource/ur5e_with_gripper.urdf) is an example in our ROS 2 packages that should be released this summer with Webots R2022b.

BenjaminHug8, 2022-02-23:
> I guess it might help users you are right, but as you can use the tool with ROS 1 (which is suppose to work with python2 until `noetic` I guess) I will add something accordingly.

I confirm that ROS 2 and ROS 1 noetic uses python3.

ROS 1 Melodic works with python2 but I believe it should also work with python3. As we no longer support Melodic I will simply change the command for `python3`.

BenjaminHug8, 2022-03-04:
@DrakeProvost is your issue solved?

DrakeProvost, 2022-03-08:
@BenjaminHug8 yes it is. Thank you for the prompt responses throughout this thread!

closed 2022-03-08 by BenjaminHug8