cyberbotics/urdf2webots #178: Converter prunes URDF branches of sensors

astumpf, 2022-06-27:
Unfortunately, I found out about this problem only recently. But latest version of the converter is pruning important (static/fixed) links from the URDF where sensors are attached are removed. In case of cameras, the optical frames are truncated. I suspect that Webot has a fallback to attach them to the root link as seen in the following screenshot.

Current behavior:

Camera frame is supposed to be the optical link that is truncated by the converter.

[image]

Expected behavior:

Cheated by changing the URDF to set the camera frame to the last non-truncated link.

[image]

TheMangalex, 2022-07-05:
The RemoveDummyLinksAndStaticBaseFlag function doesn't check if a link has sensors, as they haven't been parsed at that point. 
Additionally a preserveFixedJoint or preserveLink tag would be needed in case some frames  are needed in ros applications.

astumpf, 2022-08-08:
This issue is currently a major blocker when working with Webots. Any hint or ideas to fix it?

astumpf, 2022-10-06:
Solved with #189

closed 2022-10-06 by astumpf