cyberbotics/urdf2webots #180: Question: Update rate of sensor

TheMangalex, 2022-07-27:
In the sdf format an update_rate for sensors is defined, but unfortunately this field is not available in the parser right now. 
This field would be useful to increase or decrease sensor updates of cameras, range finders, imus, etc. and therefore useful to improve upon simulation performance and match real robots better. 
Is there a way how this can be implemented/passed to the webots ros controller?

omichel, 2022-07-27:
Do you mean this `update_rate` field is specific to the SDF format only? Or can we specify it in a URDF file as well? If yes, can you provide an example?

TheMangalex, 2022-07-27:
The update_rate field is a child of the sensor field and can be specified for sensors. 
The [ros wiki](http://wiki.ros.org/urdf/XML/sensor) says it is part of the urdf in section 2.

It seems currently not to be parsed by the parser.
An example for a camera would be:
<sensor type="camera" name="thermal_camera_sensor">
        <update_rate>9</update_rate>
        <camera>
        ...
[Source:](https://github.com/tu-darmstadt-ros-pkg/hector_models/blob/83d3ccbb2b5a9c31095c9ba4f4a5c2521ac4a29d/hector_sensors_description/urdf/flir_a65_camera.urdf.xacro#L46)

To parse these information into a proto, or do I need to create custom controllers?

omichel, 2022-07-27:
There is no such information in Webots PROTO files, so I guess it is better to create a custom controller where you will call:

```C
wb_camera_enable(update_rate);
```

TheMangalex, 2022-07-27:
Thank you for the quick responses

closed 2022-07-27 by TheMangalex