cyberbotics/urdf2webots #184: Geometry.reference database does not consider scaling option

astumpf, 2022-09-30:
Using the `Geometry.reference`, all shapes loaded from the same mesh are getting replaced even though they are mirrored (scaling factor -1.0 in one axis). This leads to wrong use as the newly introduced `ccw` flag which is part of the geometry and must be adapted based on the scaling factors.

This leads to wrong rendering as faces are pointing in the wrong direction:

### Correct faces:
```
children [
  Transform {
    scale 1.000000 -1.000000 1.000000
    children [
      DEF wheel_small_mesh_visual CadShape {
        url "<my_path>/wheel_small_mesh.obj"
        ccw FALSE
      }
    ]
  }
]
```

[image]

### Wrong faces when mirrored (wrong ccw as forced copied from original mesh object):
```
children [
  USE wheel_small_mesh_visual
]
```

[image]

astumpf, 2022-09-30:
Fix in #185

omichel, 2022-09-30:
Fixed in https://github.com/cyberbotics/urdf2webots/pull/187.

closed 2022-09-30 by omichel