cyberbotics/urdf2webots #186: Mesh of base is not rendered

astumpf, 2022-09-30:
Currently the first rigid body (base_link) is not rendered anymore. The exporter generates following code snippet:

```
Solid {
  children [
    Transform {
      translation 0.260000 0.000000 -0.200000
      children [
        DEF base_mesh_visual CadShape {
          url "<my_path>/meshes/base_mesh.obj"
        }
      ]
    }
  ]
  name "structure_link"
  boundingObject Group {
    children [
      Transform {
        translation 0.090000 0.000000 -0.080000
        children [
          Box {
             size 0.561000 0.400000 0.280000
          }
        ]
      }
      Transform {
        translation 0.426750 0.000000 -0.200000
        children [
          Box {
             size 0.112500 0.400000 0.040000
          }
        ]
      }
      Transform {
        translation -0.311750 0.000000 0.190000
        children [
          Box {
             size 0.322500 0.400000 0.340000
          }
        ]
      }
    ]
  }
  physics Physics {
    density -1
    mass 70.528000
    centerOfMass [ -0.025600 0.000000 -0.030700 ]
    inertiaMatrix [
      2.763623e+00 5.149228e+00 4.973214e+00
      -3.619958e-02 1.429088e+00 5.323388e-02
    ]
  }
}
```

However, when I replace the `shape` part with the result from an older version, the base mesh is rendered correctly:
```
children [
  Shape {
    appearance PBRAppearance {
      baseColor 0.500000 0.500000 0.500000
      transparency 0.000000
      roughness 1.000000
      metalness 0
      emissiveColor 0.000000 0.000000 0.000000
    }
    geometry DEF base_mesh Mesh {
      url "<my_path>/meshes/base_mesh.obj"
    }
  }
]
```

PS: Sorry, I cannot show any images of the full robot.

ad-daniel, 2022-09-30:
Based on the snippet and the fact you didn't mention any errors, I assume the node generation is correct, but rather the issue seems to be that the resulting shape is invisible, correct? Would it be possible to have the obj or at least the mtl? (here or via mail)

astumpf, 2022-10-05:
You got a mail :).

astumpf, 2022-10-05:
Yes, I got no errors in Webots. All other parts are visualized correctly so far I recognized. The robot is also simulated as usual.

stefaniapedrazzi, 2022-10-05:
We received your mail and we can reproduce the issue.
This is a problem specific to the OBJ import, because if I convert it to Collada DAE everything works correctly.
We will inspect it and keep you posted.

By the way I get this warning in Webots while loading the OBJ file:
```
WARNING: Solid > CadShape: Mesh 'defaultobject' has more than 100'000 vertices, it is recommended to reduce the number of vertices.
```
Following this warning I tried to remove some of the vertices so that the resulting mesh has less than 100'000 vertices and the mesh was correctly visualized in Webots.

astumpf, 2022-10-05:
Yes indeed. We generated a version with much fewer vertices and that works now. Thanks for quick your feedback!

omichel, 2022-10-06:
I guess we can now close this issue. If not, feel free to re-open.

closed 2022-10-06 by omichel