cyberbotics/urdf2webots #199: How to generate a passive joint?

Bozenton, 2023-03-11:
Hi, there

Thanks for your sharing of work! 

I want to create a passive joint for my inverted pendulum car model, where the joint connecting car with rod need to be passive. I refer to an example provided by webots: 

https://github.com/cyberbotics/webots/blob/9c6469597a29f84dff5fcf05d3443d8fd38947f4/projects/samples/howto/inverted_pendulum/worlds/inverted_pendulum.wbt#L70-L79

I find that its HingeJoint only has device of PositionSensor, but doesn't have a RotationalMotor. Is there any way to make my joint like this when converting urdf file to proto? 

Hope for your reply :smiley:

omichel, 2023-03-13:
Yes, it should be possible to do this from a URDF file and then convert it to Webots.
However, it would be simpler to create it directly in a Webots PROTO or Webots world file, taking inspiration from the example you mention.
Why don't you do that?

closed 2023-03-13 by omichel

IDavGal, 2023-09-04:
I'm interested in this as well, and cannot find how to do it using the tool. According to the [writeRobot](https://github.com/cyberbotics/urdf2webots/blob/905b12c8abdb1e3a847a0fc8d10772d1a5a0d7fe/urdf2webots/writeRobot.py#L497) code, the motor is created along with the HingeJoint. How would it be possible?

In my case I'm working in a webots simulation for a robot that's defined with a URDF and is subject to change, so I'm generating the proto on runtime.