cyberbotics/urdf2webots #200: URDFSpawner <device reference=”link”> does not work when the link is defined in the urdf and not in an existing PROTO

PymZoR, 2023-04-04:
**Describe the Bug**
When defining a robot using only one URDF and no existing PROTO, `reference` property of the `device` tag does not work.
In the turtlebot example, it works because the turtlebot already exists in the world file and the urdf is only used for defining webots plugins : https://github.com/cyberbotics/webots_ros2/blob/master/webots_ros2_turtlebot/resource/turtlebot_webots.urdf#L4

**Steps to Reproduce**
Consider the following urdf : 
```xml
<?xml version="1.0"?>
<robot name="myrobot"
    xmlns:xacro="http://ros.org/wiki/xacro">

    <!-- Webots -->
    <webots>
        <device reference="laser_link" type="Lidar">
            <ros>
                <enabled>true</enabled>
                <updateRate>5</updateRate>
                <topicName>/scan</topicName>
                <alwaysOn>true</alwaysOn>
                <frameName>laser_link</frameName>
            </ros>
        </device>

        <plugin type="webots_ros2_control::Ros2Control" />
    </webots>

    <!-- BASE LINK -->
    <link name="base_link">
        <visual>
            <origin xyz="0 0 0.075" />
            <geometry>
                <box size="0.3 0.3 0.15" />
            </geometry>
        </visual>
        <collision>
            <origin xyz="0 0 0.075" />
            <geometry>
                <box size="0.3 0.3 0.15" />
            </geometry>
        </collision>
    </link>

    <joint name="laser_link_joint" type="fixed">
        <parent link="base_link" />
        <child link="laser_link" />
        <origin xyz="0 0 1" />
    </joint>

    <!-- LINK2 -->
    <link name="laser_link">
        <!-- Here the link could be empty resulting in only a Transform Node with a Lidar object, but due to issue cyberbotics/webots_ros2#559
        a <collision> tag is required in order for the link to exists in webots -->
        <visual>
            <origin xyz="0 0 0.075" />
            <geometry>
                <box size="0.3 0.3 0.15" />
            </geometry>
        </visual>
        <collision>
            <origin xyz="0 0 0.075" />
            <geometry>
                <box size="0.3 0.3 0.15" />
            </geometry>
        </collision>
    </link>
</robot>
```

**NOTE** : I added `<collision>` and `<visual>` tags to the `laser_link` otherwise it is not created in webots. Probably linked to the issue : https://github.com/cyberbotics/webots_ros2/issues/559

The `laser_link` is not created, nor the Lidar object.

**Expected behavior**
A Lidar object should exist in Webots and the ROS ropic `/scan` should be advertised.

**Affected Packages**
List of affected packages:
  - webots_ros2_driver

**System**
 - Webots Version: R2023a for macOS
 - ROS Version: Humble
 - Operating System: Ubuntu jammy in docker, host macOS Monterey 12.2.1, using local_simulation_server.py

ygoumaz, 2023-04-04:
@omichel Is there a reason why the `<gazebo>` tag is mandatory to import sensors? Is it because `urdf2webots` was first (and only?) intended to convert gazebo URDFs to Webots PROTOs? If this is the case we may extend it to single `<sensor>` tags, as URDFs may be used in `webots_ros2` and created from scratch. On another hand, it may be good to encourage users to create their robots in PROTO format. Any thoughts?

omichel, 2023-04-11:
Yes, this `<gazebo>` tag was supported to ease the Webots conversion from URDF files designed for Gazebo.
I agree it is way better to create and maintain a robot model as a Webots PROTO file and use Webots to export the corresponding URDF model for use with ROS nodes.

closed 2023-04-21 by ygoumaz